athertop/MavLink_FrSkySPort #3: "Invalid Flight Mode"

ThomasH3, 2017-01-20:
Hello
I'm receiving a message which is constantly switching between stabilize and invalid flight mode. 
What do I need to change to display the flight mode correctly ?
It is possible to display ch7 and ch8 APM Output .... Selection could be a safety switch ... "Return to home" or "Land" or a autotune - autotrim selection.
I'm using APM mini 3.2

athertop, 2017-01-20:
The latest Teensy code creates 2 "Fuel" sensors - the one on instance 8 contains the flight mode, whilst the one on instance 9 needs to be renamed to MSG. Is it possible you didn't rename the second one here? If its not renamed, then the LUA screen will see the d uplicate sensors and not know which to use for flight mode. If you did rename the (instance 9) sensor, and are still getting the confused flight modes, then please check the Fuel sensor on your model telemetry screen and see if the number might be switching between values. I have seen one instance of this in the past and it was fixed by deleting the Fuel sensor and having it re-discovered (was a strange OpenTx quirk).
Cheers, Paul

ThomasH3, 2017-01-22:
All fixed ! ...... It works really great ...... Thanks

athertop, 2017-01-22:
Was it that you forgot to rename second Fuel sensor to MSG?

closed 2017-01-22 by athertop