athertop/MavLink_FrSkySPort #4: Batt2 Sensor

Juzzle1, 2017-02-03:
Would it be possible get sensor data from the second battery (BATT2)? On recent versions of Copter / Plane this is sent over Mavlink just like Battery 1.

I only need the additional sensor data to be sent to the Taranis (no need for it on LUA screen), but Pixhawk 2 users will likely need this too as it has two current/voltage sensor inputs for dual battery setups.

athertop, 2017-02-03:
I'll take a look at this. Should be possible,  but will require you to turn on this functionality as an option in the addons/options section of the teensy code once I have added the new code. May take a few weeks as I'm on vacation just now.

athertop, 2017-02-19:
@Juzzle1 - Can I please verify that in addition to using two battery monitors, you do not also have installed OpenTx FLVSS (or any other battery) sensors? The reason I ask, is that I plan to use the telemetry sensor ID 15 for the BATT2 device (which is also used by a second FLVSS sensor should one be installed). So using BATT2 you would not be able to also have  FLVSS FrsSky sensors configured. Thanks

Juzzle1, 2017-02-19:
Thanks for looking into this. 

No other sensors in use here, just Pixhawk -> Teensy -> S.Port

athertop, 2017-02-27:
@Juzzle1 - I have just uploaded changed Teensy code to https://github.com/athertop/MavLink_FrSkySPort/tree/Mav_Sport2.1-2.2-dev/Teensy/MavLink_FrSkySPort (this is the -dev branch) which includes a BATT2 feature.
This must be enabled under features section in
MavLink_FrSkySPort.ino. Once discovered this should create a second (duplicate) VFAS sensor but using the sensor instance 15 (check the sensor IDs to locate this one). This must be renamed by editing the discovered sensor to keep the name unique. Call it say BAT2 if you like.
This has been tested for compilation but I am not able to test the functionality in practise as I don't have a Pixhawk 2.1 with 2 power bricks. Also note that although the teensy code is updated here to create the new sensor, this does not yet appear on the LUA Telemetry screen. I will sort this later for you if you need it. If you can please test this.
Some notes on limitations in its use: if the USE_BATT2 feature is enabled then you can not have the feature USE_SINGLE_CELL_MONITOR or USE_FLVSS_FAKE_SENSOR_DATA enabled if MAXCELLS is set to >6 - as this would create another sensor also using ID15 so a clash.

Juzzle1, 2017-02-27:
Hi Paul,

This sounds great. 

FYI, I'm actually using this with a Pixhawk 1, with BAT2 currently configured to use the 6.6V ADC which is reading the voltage from a Mauch Sensor Hub relay output. 
I have 2 batteries connected and the sensor hub outputs 5V when both batteries are discharging with a similar current, and 0V when there is >20% difference in discharge rate (e.g. when one battery is not connected or is at a different charge level).

So as long as I can get BAT2 values down to the Taranis (right now I can only see the value in mission planner), I can set up an audible alert when BAT2 = 0V (fault condition).

I'm using my aircraft tomorrow morning, so I'm not going to fiddle with / break anything tonight, but will have a go tomorrow evening if I have time.

Thanks so much for doing this!

Justin

athertop, 2017-02-27:
Just checked out the sensor hub - first time I've seen this configuration. Looks awfully complex for what essentially us 2 Lipos in parallel. I do the same but lipo sensing done differently. For me, both 4s Lipos are wired to the input of one apm battery sensor - I am actually using a 180a attopilot version,feeding v/a in usual way to pins on the power connector of the pixhawk. The power is provided to the pixhawk by a 5v 5a BEC. Then I make use of  two FrSky Flvss lipo sensors, each monitoring the 4 cells of its lipo via the LiPo balance connector. I have these configured with different sensor id's, and both show up on the Taranis where I have configured calculated sensors to give me the Cmin value for the telemetry screen. The attopilot provides v/a to the Pixhawk which appears as VFAS/Curr in the Frsky telemetry.

Juzzle1, 2017-03-05:
Sorry I've not had a chance to test this yet. Hopefully I'll have some time tonight or tomorrow. I'll keep you posted.

I went for the Mauch sensors as they use much more accurate hall effect sensors and the manufacturer calibrates them before shipping so you can just plug the provided voltage / current divider values into MP and get accurate readings without fuss. 

Cheers

Juzzle1, 2017-03-06:
Just tried 2.2-dev with USE_BATT2 enabled, ran sensor discovery and hey presto, I can see the BAT2 voltage reading. 

From what I can see, everything else seems to be working normally as before.

I'll take it out for a proper spin at the weekend (weather permitting), but testing on the bench looks good.

Thanks for putting the time in on this, much appreciated.

Justin

athertop, 2018-07-09:
BATT2 was moved to the current master a while ago. Closing

closed 2018-07-09 by athertop