athertop/MavLink_FrSkySPort #5: Telemetry screen bit off

Dutchy-79, 2017-02-16:
Hi,

I'd better continue here because I just switched to your fork,
Haven't tried the sounds yet but I will once my last battery is charged.

Just did a hover in the garden to tune and noticed the telemetry screen C-Min mAh and Wh are a bit off, this also happened in the Cloony82 master.
[image]

It's on the wrong side if i compare to your screenshots.

Regards,

Jasper

athertop, 2017-02-16:
I'm wondering here, you must be using OpenTx 2.2? I have noticed the same offsets - as 2.2 is still in a state of  development, each new recent release candidate has changed numeric alignment methods, so it's making it difficult to keep up. Hopefully this new rc10 version is close to being final, but I think one of the suggested alignment changes I was discussing with the Opentx des didn't make the rc10 release so this may change again for the next candidate version.  I will look at putting in a temporary alignment fix in my - dev branch tonight if I find the time but be aware that the heat release may change the alignment again.

Dutchy-79, 2017-02-16:
Ahh thanks, that explains.

O noticed it on the right on the meters from home also. 

And still no sounds when returning to home on empty battery or reaching fence. 

I might try to go back to 2.1.9 if I find some time before going to work tonight. 

If 2.2RC10 is the problem, or is there a older 2.2 that worked already?

athertop, 2017-02-16:
I would suggest leaving it on rc10 and I will address the alignment issues. As for the voice messages, these should work regardless of the Opentx version (2.1 or 2.2). I have no way to test these as no models on apm 3.21 any more. Make sure the fuel sensor on id 8 is reflecting the changes in flight mode. RTL is just a flight mode after all. 
Edit - I just determined that the sounds provided for 3.2 only cover flight modes for copter and not plane. I will add the plane flight mode files tonight also to the dev repo. If you are using copter, then you should be fine for sounds, but not sure if you are maybe using plane? 
Incidentally, to choose between copter/plane on my new 2.0 version of the lua scripts, you will need to select between these on the telemetry screen config page as per wiki instructions. I presume you are using the v2.0 scripts?

Dutchy-79, 2017-02-16:
Thanks, I'm using copter. I can see when it returns on the left top off the telemetry screen so I'm pretty sure the messages come back like they should. I will copy all again to the SD card, maybe I did something wrong.

Dutchy-79, 2017-02-16:
Sounds work, I uploaded the complete folder so not deleted the AC3.2 part. 

One more question, why is TxBat 0.0 and how to set it up?

athertop, 2017-02-16:
That's a very good question - I have traced it to the fact that the LUA code which displays this value is referring to the source by a value which has been changed in OpenTx 2.2. This should be simple enough to resolve. I'll add it to the list. I didn't get chance tonight to rectify the screen layout, but will look at this over the weekend. Thanks, Paul

Dutchy-79, 2017-02-16:
No hurry, sounds work and for the TxBat I'm getting the standard sound notification already. 

Working night shift till Saturday so won't be flying that much anyway.

athertop, 2017-02-16:
OK, so hopefully you will be glad to hear that I have fixed both the alignment and TxBat voltage readings in my local lua file here and have tested the fixes on both OpenTx 2.1 and 2.2 successfully. I will need some time to compile the files and release the updates so maybe this time tomorrow. 
to resolve both issues there will be just one replacement file - mainrun.lib so pretty easy to implement the fixes for you. Will update tomorrow. Cheers, P

athertop, 2017-02-17:
Dutchy, I have just uploaded the fix for the screen layout to my -dev branch. You can download the new file directly from here: https://github.com/athertop/MavLink_FrSkySPort/raw/Mav_Sport2.1-2.2-dev/Taranis/LUA_SD_Card/SCRIPTS/TELEMETRY/LIBRARY/mainrun.lib

Just replace your original file inside the LIBRARY folder on your SD card with this new one and hopefully all will look perfect again. Please let me know how you get on.

Dutchy-79, 2017-02-18:
Perfect, as expected. 

Great work, also sounds are great since I put them in the right folder.

closed 2017-02-18 by Dutchy-79

Dutchy-79, 2017-02-21:
@athertop Just got a tip on rcgroups that you can let the lua look at a timer, so when that timer resets (for instance Timer 1 for recording flight time) then the lua script will also reset.

If this is really possible then you can reset the battery monitor with the reset flight function in the Taranis and don't have to power cycle after each battery.

athertop, 2017-02-21:
Of course I know you can check the value of the timer, and reset the Wh to zero if it sees the timer at zero, but I am thinking it would be a little more complex that at first glance. For instance, Wh is not your only consumption counter, you have mAh also - and this value is not held in any LUA variable, but in the Taranis telemetry (calculated sensor), so I would have to get the script to reset the telemetry also. I would also have to put in some mechanism so that it only does this once (say a flag to say its been reset), otherwise every time it cycles the loop it will reset the telemetry and that can be several times a second. Let me have a look anyway and see what I can come up with.

athertop, 2017-02-21:
@Dutchy-79 , I have implemented the suggested script changes and it tests out perfectly on Companion Taranis simulator. Even though I have it working, I think I could make a better functioning solution. Let me explain -
I have the script currently resetting the Wh value to zero, if:
1. it detects timer 1 has been set to zero, 
and
2. it has to also detect a battReset flag has been set to FALSE.

Let me explain that battReset flag - If I didn't have a flag in operation here (so imagine it just resets the Wh when the timer is zero), then imagine when you first switch on your radio, and the telemetry script launches. At this point, the script creates timer1 and sets it to zero, and whilst your craft remains disarmed, that timer stays set to zero - it could be for 5-10 mins if you are messing with your model. Now all the time this timer is set to 0 the script would be continually resetting the Wh counter to zero - maybe a thousand times a second. So to avoid that:
1. the battReset flag is set to TRUE when the script launches
2. you arm your craft, the timer starts counting up (so no longer zero)
3. the battReset flag is set to FALSE
...then you go flying
4. at the end of the flight, you disarm, and the scrip then does those 2 checks I mentioned earlier. It sees the timer is above 0 and the battReset flag is FALSE, so it does nothing.
5. you then reset the flight from the menu, which causes Timer1 to be reset to zero (and incidentally, the mAh value to be reset to zero also)
6. Now both conditions are met so the script sets the Wh value to zero - and then it sets the battReset flag to TRUE (which stops that continual reset from happening again as explained above).

So this all works fine, but I would rather have the Wh value resetting to zero  when the script detects that you have selected the 'Reset Flight' menu option as this is a cleaner solution withoutb the need for all those safeguards, so I have asked some questions to my OpenTx gurus on RCG to see if they can tell me how I would detect the reset flight option had been chosen. Watch this space!

Dutchy-79, 2017-02-22:
Wow, nice work m8

If you want me to test let me know.

athertop, 2017-02-22:
@Dutchy-79, I have uploaded the amended main.lua script to the -dev branch of the repo here:

https://github.com/athertop/MavLink_FrSkySPort/tree/Mav_Sport2.1-2.2-dev/Taranis/LUA_SD_Card/SCRIPTS/TELEMETRY
Please feel free to give it a try and let me know how you get on. Cheers, Paul

athertop, 2017-02-23:
I tested this today and it works fine. I have also working on another new feature which allows you to output a high channel value on a channel of choice when the craft is armed (the choice of channels 9-14 can be selected on the config screen to line up with the aux ports of the Pixhawk) - I will be using it to turn on a strobe beacon on my LRS wing when it is armed.

Dutchy-79, 2017-02-26:
Hi,

Took a while before I could go out and try, It works great now. Thanks

athertop, 2017-02-26:
That's great. Thanks for letting me know. Just looking at the next mod which has been requested for second power brick (for pixhawk 2.1). Not quite sure how I will test it since I don't have the new ph myself!

Dutchy-79, 2017-02-26:
Hmm, can't help you with that. Only have APM