athertop/MavLink_FrSkySPort #8: Wh limit

Juzzle1, 2017-02-24:
Hi Paul,

I know you recently increased the Wh limit from 250 to 500, but is there any reason this couldn't go all the way up to 999 or more for bigger aircraft?  

I'm using 2 x 12Ah 6S packs which work out at almost 550Wh. I can get by with 500 as a maximum as it's pretty close, but it would be nice to have the correct value in there.

What would happen if I just updated the value stored in SCRIPTS\TELEMETRY\DATA\modelname.txt rather than adjusting on the LUA screen - would that break something?

Cheers,

Justin

athertop, 2017-02-24:
Sure, I can up it to 999. You can try updating the value in the cfg  file for a quick fix if you like.

Juzzle1, 2017-02-24:
OK great - I'll change the config file for now.

Thanks!

Juzzle1, 2017-02-24:
I've just noticed on the config screen that I also can't select mph (it just cycles through m/s, kph and kts).
Do you have the same issue or is it related to my setup?

athertop, 2017-02-24:
Yes I figured that out yesterday and have a fixed version ready to upload. I'll sort the Whole setting before I do. The new version also includes the ability to select a channel to output high pwm value on when the craft is armed. I will use it to start the strobe beacon on my LRS plane.

Juzzle1, 2017-02-24:
Sounds like a great feature - will there be an option to set PWM high for a number of seconds or will it simply stay high whenever it's armed?

athertop, 2017-02-24:
I am thinking of re-jigging it, so instead of affecting a channel directly, I will get it to (configure and) turn on a logical switch (of choice), and you can then configure other logical switches based on this switch to configure a timed switch, such as by using the edge function. So you can then use this to effect a channel, or do whatever you want with it. The other option is to get it to set a value on a global variable which you can then use as the source for a channel. Interested to hear your preference.

athertop, 2017-02-24:
I have uploaded the amended lua code to the -dev repo for you to try out. Can I please ask that you replace both the main.lua, pls the LIBRARY folder and its contents as all the files have changed as part of my ArmGvar mod - I decided on setting te selected Global Variable to 1024 when the craft is armed. This can be disabled by setting the ArmGvar value to zero. Thanks, Paul
Repo is here: https://github.com/athertop/MavLink_FrSkySPort/tree/Mav_Sport2.1-2.2-dev/Taranis/LUA_SD_Card/SCRIPTS/TELEMETRY

Juzzle1, 2017-02-25:
I set up a simple logical switch triggered when GV1>20 and it worked perfectly.

Think I'll use this to switch off some power draining 12v accessories when disarmed eg lights and vtx and configure with a manual override switch.

I was also able to change the Wh setting above 500. 

I'll do some more testing when I have time but looking great so far.

Justin

athertop, 2017-02-25:
I think I will change the GV value to 100 instead of 1024. This way it can be used as the offset for a channel configured in the mixer with the MAX input - I.e. 100 being full stick deflection. Will still work with your logical switch setup. >20 so all good! Thanks for testing. I will move it to the master branch shortly.

athertop, 2017-02-25:
Have merged the changes into Master, so will now close this issue. Thanks again @Juzzle1

closed 2017-02-25 by athertop