athertop/MavLink_FrSkySPort #11: Support APM native (passthrough) FrSky Telemetry

Clooney82, 2017-05-01:
Hi Paul,

is it possible to add support for native (passthrough) FrSky Telemetry in LUA Telemetry Script?

BR, Jochen

athertop, 2017-05-01:
Hey Jochen, Nice to hear from you. Hope you aprove of the changes to the project since you moved on. Your work is a big inspiration to me.
Yes, it is possible to change the LUA to work with passthrough, although its beyond my scripting capabilities - but I'm sure within your own.
I was given an excel sheet (Thanks to Luis Vale) detailing the passthrough protocol - its a bit complex, as some of the sensors carry multiple sensor data on a time sharing basis. You'll see what I mean if you check out the excel sheet. Here's a link to a copy on my dropbox: https://www.dropbox.com/s/um5n2iuvriwlqh2/option10_APM.xlsx?dl=0

Wondering if perhaps you could assist with another issue I have little idea of? In some cases the arrow head in the radar display is presented outside of the square radar window. I managed to simulate this in the Companion simulator for Taranis, by setting GPS lat/long, setting the heading, and then setting Tmp2 to 1 to simulate arming, then adjusting the lat/long and heading to simulate copter movement. I am struggling to understand the maths as to how that arrow is positioned inside the radar window. Wonder if you could possibly advise to determine how the arrow can be more closely restricted to being inside that radar window?

lvale, 2017-05-02:
I'm still waiting for the release version of OpenTX to come to then start any kind of scripting on the radio

Clooney82, 2017-05-10:
Hi Paul,
I think I managed to spilt the values again (currently untested). 
And I think I found a bug in ardupilot code (see: https://github.com/ArduPilot/ardupilot/issues/6221) for passthrough Telemetry.
Next step would be to write a test lua screen to verify this on my taranis. Maybe I find time during the next days to do this.
Not all code posted below will work on a taranis, but I just wanted to keep you up-to-date.

Cheers

--------------------
-- ID: 5000 -- Text Messages  --
--------------------
```
message_value = 0x434F4D50
-- message_value = getValue("5000")

char_3 = bit32.extract(message_value,  0,  7)
char_2 = bit32.extract(message_value,  8,  7)
char_1 = bit32.extract(message_value, 16,  7)
char_0 = bit32.extract(message_value, 24,  7)
sev_1  = bit32.extract(message_value,  7,  1) -- 0x1
sev_2  = bit32.extract(message_value, 15,  1) -- 0x2
sev_3  = bit32.extract(message_value, 23,  1) -- 0x4

sev_2  = bit32.lshift(sev_2,1)
sev_3  = bit32.lshift(sev_3,2)

sev    = bit32.bor(sev_1, sev_2, sev_3)
print(sev)
print(string.char(char_0))
print(string.char(char_1))
print(string.char(char_2))
print(string.char(char_3))

```

--------------------
-- ID: 5001 -- AP_STATUS        --
--------------------
```
ap_status_value = 0x87654321
-- ap_status_value = getValue("5001")

fmod    = bit32.extract(ap_status_value,  0,  5)
ssimple = bit32.extract(ap_status_value,  5,  2)
landed  = bit32.extract(ap_status_value,  7,  1)
armed   = bit32.extract(ap_status_value,  8,  1)
bat_fs  = bit32.extract(ap_status_value,  9,  1)
ekf_bad = bit32.extract(ap_status_value, 10,  1)
```

--------------------
-- ID: 5002 -- GPS            --
--------------------
```
gps_value = 0x01234567
--gps_value = getValue("5002")

sat_num      = bit32.extract(gps_value,  0,  4)
fix_type     = bit32.extract(gps_value,  4,  2)
hdop_dm      = bit32.extract(gps_value,  6,  7)
hdop_potz    = bit32.extract(gps_value, 13,  1)
adv_fix_stat = bit32.extract(gps_value, 14,  2)
alt_dm       = bit32.extract(gps_value, 22,  1)
alt_potz     = bit32.extract(gps_value, 29,  1)
alt_p_m      = bit32.extract(gps_value, 31,  1)

alt_m  = 10^alt_potz * (alt_dm/10)
if alt_p_m then
  alt_m = alt_m * -1
end
hdop_m = 10^hdop_potz * (hdop_dm/10)
```

--------------------
-- ID: 5003 -- Battery        --
--------------------
```
batt_value = 0xFFFFAFFF
--batt_value = getValue("5003")
--print(batt_value)

VOLTAGE = bit32.extract(batt_value, 0,9)
CURRENT = bit32.extract(batt_value, 9,7)
CURRENT_POTZ = bit32.extract(batt_value, 16,1)
TOTALMAH = bit32.rshift(batt_value, 17)

print(TOTALMAH)
CURR = 10^CURRENT_POTZ * CURRENT
print(CURR)

print(VOLTAGE/10)
```

--------------------
-- ID: 5004 -- home           --
--------------------
```
home_value = 0x00000000
-- home_value = getValue("5004")

dist_home_m    = bit32.extract(velandyaw_value,  0, 10)
dist_home_potz = bit32.extract(velandyaw_value, 10,  2)
alt_home_m     = bit32.extract(velandyaw_value, 12, 10)
alt_home_potz  = bit32.extract(velandyaw_value, 22,  2)
alt_home_p_m   = bit32.extract(velandyaw_value, 24,  1)
angle_home_deg = bit32.extract(velandyaw_value, 25,  7) / 0.00333

dist_from_home = 10^dist_home_potz * dist_home_m
alt_from_home  = 10^alt_home_potz * alt_home_m
if alt_home_p_m then
  alt_from_home = alt_from_home * -1
end
```

--------------------
-- ID: 5005 -- vel and yaw    --
--------------------
```
velandyaw_value = 0x01AC02AC
-- velandyaw_value = getValue("5005")

vert_vel_dm_s = bit32.extract(velandyaw_value,  0,  7)
vert_vel_potz = bit32.extract(velandyaw_value,  7,  1)
vert_ver_p_m  = bit32.extract(velandyaw_value,  8,  1)
hor_vel_dm_s  = bit32.extract(velandyaw_value,  9,  7)
hor_vel_potz  = bit32.extract(velandyaw_value, 16,  1)
yaw_degree    = bit32.extract(velandyaw_value, 17, 11) / 0.05 -- 0.2 degree

vert_vel_m_s = 10^vert_vel_potz * (vert_vel_dm_s / 10)
hor_vel_m_s  = 10^hor_vel_potz  * (hor_vel_dm_s  / 10)
```

--------------------
-- ID: 5006 -- ATTI AND RNG   --
--------------------
```
attiandrng_value = 0x00FF0003
--attiandrng_value = getValue("5006")
--print(attiandrng_value)

roll_degree  = bit32.extract(attiandrng_value,  0, 11) / 0.05
pitch_degree = bit32.extract(attiandrng_value, 11, 10) / 0.05
rng_alt_cm   = bit32.extract(attiandrng_value, 21, 10)
rng_alt_potz = bit32.extract(attiandrng_value, 31,  1)
alt_cm = 10^rng_alt_potz * rng_alt_cm
```

--------------------
-- ID: 5007 -- Param          --
--------------------
```
param_value = 0x00FF0003
--param_value = getValue("5007")
--print(param_value)

paramID = bit32.extract(param_value, 0, 8)
paramValue = bit32.rshift(param_value, 8)

--print(paramID)
if paramID == 1 then
  print("MAV_Type")
else 
  if paramID == 2 then
    print("Batt Voltage")
    paramValue = paramValue / 100.0
  else 
    if paramID == 3 then
      print("Batt FS mAh")
    else
      if paramID == 4 then
        print("Batt Capacity configured (mAh)")
      end
    end
  end
end

print(paramValue)
```

Clooney82, 2017-05-10:
Do I need opentx 2.2 or does it also work with 2.1.9?

athertop, 2017-05-10:
You'll need 2.2 :-)

Clooney82, 2017-05-10:
Ok, so I will try with 2.2.
Cause on 2.1.9 is see all Sensors, but Script doesn't displays values.

athertop, 2017-05-10:
I think the sensors will be hidden on 2.2, but should be accessible using the sensor numbers. This is all of course based on what I have read as haven't tested this myself. Keen to know how you get on though so keep us posted!

Clooney82, 2017-05-11:
Today I testet 2.2 but I went back to 2.1.9, because no telemetry is working for me.
Maybe I try again later.
But I made a script for passthrough telemetry. 
It runs on mine without errors, but does not display any telemetry data, because it is not working on my TX.
[test.lua.txt](https://github.com/athertop/MavLink_FrSkySPort/files/992637/test.lua.txt)

athertop, 2017-05-11:
Jochen, I believe that the sensors are invisible on 2.2 by design, so you don't need to discover them. Maybe an idea to check out the instructions in the Craft & Theory website for more accurate info.

Clooney82, 2017-05-23:
found something in the deep of the internet ;)
You need to replace getValue(xyz) with some other code, because this is not supported for passthrough telemetry.
*read passthrough telemetry with sportTelemetryPop()
*create own Telemetry Sensor with setTelemetryValue(id, subID, instance, value [, unit] [, precision [, name]) - not really necessary (maybe if you need telemetry logging)

e.g.
```
local function bg_process_passthrough()
	local i0,i1,i2,v
	i0,i1,i2,v = sportTelemetryPop()
	-- only for SIMULATOR NEEDED
	if v == nil then
		v = 0
	end
	-- END SIM PART
	-- unpack packet 5006 - Messages
	if i2 == 20486 then
		data.roll_degree   = (bit32.extract(v,  0, 11) -900) * 0.2
		data.pitch_degree  = (bit32.extract(v, 11, 10) -450) * 0.2
		local rng_alt_cm   = bit32.extract(v, 21, 10)
		local rng_alt_potz = bit32.extract(v, 31,  1)
		data.rng_alt_cm    = 10^rng_alt_potz * rng_alt_cm
		--setTelemetryValue(id, subID, instance, value [, unit] [, precision [, name])
		setTelemetryValue (5006,0,1,v,20,0,"ROLL")
		setTelemetryValue (5006,0,2,v,20,0,"PITCH")
	end
	...
	...
end
```

now some fine tuning for the values are needed. basically it works.

[MFrSP.lua.txt](https://github.com/athertop/MavLink_FrSkySPort/files/1022248/MFrSP.lua.txt)

geofrancis, 2017-06-26:
was there any more progress on this ?

athertop, 2017-06-26:
Hey George, I sent you the updated files already which as you know have a bug. I'll look at this once I have my new quad built (and so can test it in action). Might be a month, as reliant on China post!

Clooney82, 2017-06-27:
Hi Paul,
Didn't got any mail till now.
But it is no problem, cause I'm currently in vacation.

athertop, 2017-06-27:
Jochen, I was writing to George in that comment and not your good self. George is our resident Scottish drone genius ;-) he's keen to see this working. Have fun on vacation

Johnex, 2017-07-04:
If you guys need help testing i would be happy to do it!
I'm looking forward to using a Lua script for passthrough telemetry :)

lucamax, 2017-12-15:
I'm looking too for a DIY solution for passtrough telemetry .
Do someone knows what kind of hardware device is needed , i.e. as in the craft and theory cables ?

Great work for this latest release athertop, especially with the plane/copter configuration   :)
The only problem I still have  is the  txtmsg  that is often corrupted and almost unreadable.
IMO you should provide a Donate button somewhere in the Wiki .

athertop, 2017-12-15:
The hardware is simply a signal inverter and diode. To be more precise, this is exactly what the C&T adaptor uses (a TTL<->RS232 converter, plus a diode will do the trick): https://www.aliexpress.com/item/10pcs-Serial-Port-For-Arduino-MCU-Mini-RS232-to-TTL-Converter-Adaptor-Board-Module-MAX3232-3/32556620038.html

lvale, 2017-12-16:
[image]
20c worth of materials

lvale, 2017-12-16:
[image]

jplopezll, 2017-12-22:
If you are still interested in S.Port passthrough telemetry, I am writing a lua script to decode most of the available information and "paint" it in a nice screen with even a basic PFD, GPS data, messages, etc.

I am still debugging it, but the basic functionality is already implemented and working in my X9D+. I am attaching a screenshot taken from the Companion simulator taken just yesterday.

As interface I am provisionally using too optocouplers (instead of NPN transistors), some resistors and a diode. I have ordered one module like the one in the last picture above as the final option for my quad.

[image]

lucamax, 2017-12-22:
Nice work jplopezll  !
About the delay,  4 seconds it is really a lot. Is it the same for all sensors ?

jplopezll, 2017-12-23:
Thanks for your kind comment! 
Yes, the delay is the same for all sensors and even with less data being unpacked. I am now working on moving the passthrough queue pop to a custom (mix) script to see if I get better results.
I will let you know.

jplopezll, 2017-12-23:
Updating the queue part to a mix sript worsens the situation.
I have done a very simple script with only I think that the problem is with the drawing part being to heavy for the limitted computing capacity. With only pitch and roll decoding and only drawing some text it is almost instant.
I will try to check now what is the part of my original code burdening the speed...

lvale, 2017-12-23:
@jplopezll Check on the ArduPilot source code library where the 5000 range is generated the frequency they are sent down and try to adjust your decoding code to that, and only redraw in sync with new received values

jplopezll, 2017-12-24:
I'll try to reduce the drawing load and revert.

jplopezll, 2017-12-26:
I have rewritten most of the drawing code. Still debugging and adjusting the layout, but I am now able to have around 100 ms total processing time. The PFD is now moving at a rate similar to the one I can see in Mission Planner.
I don't know if anybody is still interested on the code to get passthrough telemetry data.
A screenshot of the current status of the display as shown in companion:
[image]

lucamax, 2017-12-26:
Great result jplopezll  !
Guess that everybody is interested in Frsky passtrough telemetry and his code , it do not require the Teensy and has much more data compared to Frsky native telemetry where Airspeed is missing an important data for plane pilots.

lvale, 2017-12-26:
@jplopezll I would suggest that you submit a branch from this repo. I'm sure @athertop won't mind.

jplopezll, 2017-12-27:
OK, I will do some code clean up and try to create a new branch so that you can see and improve the code.

Johnex, 2017-12-27:
@jplopezll looks super sexy 👍

jplopezll, 2017-12-28:
I have tried to open a new branch, but I have no permission to do so. I have decided to open a new repository so that I do not create a fuss in this one...
If you are interested, you can check the script here: https://github.com/jplopezll/OpenTX_FrSkySPort_passthrough_0.0.1-dev
I you think it would be better merging all here, I am open to that.

jplopezll, 2018-01-24:
I have just uploaded a fully functional telemetry script using S.Port passthrough: https://github.com/jplopezll/OpenTX_FrSkySPort_passthrough_0.0.1-dev
It is responding instantly and able to handle the huge volume of data being transmitted. I have added a scrollabe FIFO buffer for up to 10 messages (I think it is enough).
I would like you to give me some feedback if you have the chance to try it in your setups.

athertop, 2018-06-12:
Guys, The idea to get the teensy providing FrSky passthru telemetry has been realised! I have been testing [this awesome teensy code project](https://github.com/zs6buj/MavlinkToPassthru) recently which does exactly that, with [this awesome LUA telemetry screen project](https://github.com/yaapu/FrskyTelemetryScript). I think with these talented guys around, its time for me to step down here and revert to being a user again. I'd seriously recommend you checking out these two projects which work very well together. If you just want the telemetry over X series FrSky Rx then the LUA script linked above will do it nicely, but if you want to do this over ULRS (or some other remote serial telemetry modem), allowing both mavlink and the LUA script to be available on the ground, then introduce the teensy into the mix and you're all set. Going to close this one now.

closed 2018-06-12 by athertop