athertop/MavLink_FrSkySPort #12: Can you add connectivity details (teensy?) on your wiki?

marcmerlin, 2017-05-15:
Thanks for putting this up to date page and your wiki page explaining how to install the firmware.
However, I've found nothing on how to connect the LRS JR module to the Taranis.
I have the DTF-UHF JR module with inverted serial support. Does this mean it can just send the data directly to my Taranis through the port it's already connected on?
Whereas an Orange JR module needs some hacking?
I've read that a teensy is required to convert from serial to sport, but not found anywhere how to do this or how it's connected even.
Could you add this to your wiki, even just links to existing pages?
Thanks :)

athertop, 2017-05-15:
Hi Marc, The Teensy project here was adapted by me specifically to work with the ULRS system, as that is what I use myself on my big fx-61 flying wing for LRS. What you have to bear in mind is that the ULRS project is quite "DIY", and most ULRS users build their own Tx/Rx modules, but as you mention some do use (and the original direction of the ULRS project was in fact based on) the Orange LRS JR modules. I personally use the sponge board, mounted inside a project box to which a JR module is fixed to the back (just the plastic case in my design with the internal connector glued in place). I simply use the connector in the JR module to gain connection with the pins inside the module bay - so from this connector, I have the wires feeding into my project box where they provide Battery voltage from the Taranis (via a small 5v regulator) to power the ULRS mini sponge board, and all the other components such as the BT module and the Teensy board. Those connections also give me access to the S.Port connection in the module bay which connects to the output from the Teensy board. As for wiring the teensy into your type of JR module, I honestly don't have any idea how you do this as I don't actually have or know of that module to which you refer. I do have a couple of the Orange Rx Futaba modules which Marc Dornan (in the ULRS forum) kindly sent me, when my original Wolfbox JR module smoked! Obviously both the Wolfbox unit and the Futaba module are different than your own JR module, but I would say that you simply need access to that bottom (S.Port) pin in the module bay, and this access needs to be dedicated to the Teensy output (plus a ground wire also for reference), so if there are tracks on that board feeding that pin already, you will likely have to cut them (no idea of the complexities of this on your own type of JR module board, or even if ULRS runs on your type of board). Then you will have to connect the ULRS com port output to the input of the Teensy to receive the Mavlink stream. As for your module having an inbuilt inverter, that will not be used as the Teensy provides the signal in the correct format for SmartPort directly. Like I said, I have no idea if the DTF-UHF JR module is actually supported by the ULRS firmware, so I would suggest making sure that is the case before butchering your hardware to make it work with the Teensy. For info on the Taranis connections, here is a picture I found on google images: 
[image]

closed 2017-05-15 by athertop

marcmerlin, 2017-05-16:
Please accept my apologies, I just found out that I never saw a bunch of your documentation (including the page you just linked to) because my browser wasn't wide enough and I couldn't see the table of contents on the right side.
I see it now, and it should answer all my questions. Thanks again.

marcmerlin, 2017-05-23:
Thanks for the details, and sorry to have engaged you (unknowingly) in 2 different places (I'm reading your rcgroups post and writing an answer there).
The information you gave is helpful. To answer your question I have the hawkeye DTF UHF deluxe JR TX: https://www.fpvmodel.com/hawkeye-dtf-uhf-deluxetx-transmitter-taranis-jr-turnigy-compatible-djrtx-_g706.html
Because it supports sending telemetry for 2 analog devices (only) via openrlsng, it means that it must be connected to the sport pin, so I'd have to cut the trace as you said, which I can do, but than again my TX is not and will not be supported by uLRS and flipflap will not open his code to potentially allow others to contribute drivers he's not interested in writing himself, so I'd have to throw away my perfectly good working module and get another one if I switch to uLRS.
That said, I *think* I should be able to get the gitsly branch of openlrsng working on my board, and then connect your teensy code to it, as if it were uLRS and hopefully the rest will work.
Depending on how big your code it, because my JR module actually supports inverted serial and is able to generate telemetry for my taranis right now, I  think in theory your code could be merged directly into openlrsng and feed mavlink telemetry into sport instead of feeding the analog data received via the radio link, without the need for that extra teensy.
The only problem is that probably the 328p arduino on the TX module may be too small to host and run your code, making the 2nd arduino-like board a requirement anyway.
Anyway, I'll read more to make sure I understand all the details, and reply more on rcgroups.
Thanks again.

athertop, 2017-05-23:
Marc, No worries bud. I would suggest keeping on at FlipFlap (Benjuit) to see if he would support your devices - assuming they are 1W Tx/Rx he may be keen to do that but - he would need a set to make that possible. As for the onboard 328p - the code for the teensy will not run on the 328p - it did once, but then the arduino pro mini board was dropped in favour of the 32bit processing power of the teensy - many of the commands used in the code are teensy specific, so I think it would be a huge undertaking to get it to run on your Tx. There is however another similar telemetry project which does run on a mini board, so you may want to take a look at that. Link here: https://github.com/MikeBland/APM_Mavlink_to_FrSky

Best wishes on getting that integrated. Cheers, Paul

marcmerlin, 2017-05-23:
Just FYI, flipflap has already stated he will not support that TX because he doesn't like the hardware, never mind countless ones were sold to most everyone using openlrsng, who now cannot switch to uLRS without changing hardware. It's obviously his choice, but because uLRS is closed source, it also prevents someone else from supporting it.
On the RX issue, I have a better pre-built module than all the uLRS modules that exist (except for the lack of 1W backlink, it's only 100mW), a brotronics diversity RX with lost model function, backup lipo, and built in buzzer support (plus diversity antennas), but he said over a year ago he'd be interested in supporting them, but hasn't yet, likely due to lack of time. This means that I really shouldn't wait for that to happen soon.
If I'm going to have an external converter board, there is 0 reason not to use your teensy solution over a lesser chip. I had just a happy dream where one day your code could be built directly into the TX module, but clearly right now the 328p flash (and likely RAM) are full already, so it won't fit.
We can followup there though, so that others can benefit: https://www.rcgroups.com/forums/showpost.php?p=37578335&postcount=10628

marcmerlin, 2017-05-24:
As mentioned on https://www.rcgroups.com/forums/showpost.php?p=37585612&postcount=10658 , I agree we should move outside of the uLRS forum so as not to spam.
Hopefully this is an ok place, at least it's indexable/findable for others :)
If you'd rather another place/way, let me know.

You said: https://github.com/ArduPilot/ardupilot/issues/6296  "Good luck with that, although not sure why you would need this if your ardupilot already has that sensor data due to all the pixhawk/apm integrated and connected devices." => if you look at that request more closely you'll see it's specifically about the lipo cell tester, which AP does not have, and what I care even more about: RPM sensor and temperature sensors, which AP does not have either (it has some kind of RPM feature, but it does not work for most setups).

"The pixhawk would output a mavlink serial data rate of 19200 to the port in a ULRS running rx (ULRS sets the port to 19200 for rx), and your tx running ULRS would output a data rate of 57600 into both a BT module (tx and rx from rx connected) and into teensy (tx only connected)" => mmh, ok, but the mavlink data would still only flow at 19200 anyway, even if you have 57.6k going to the BT module. I suppose you do this because the BT module wants to run at 57.6k, and if so, that's fine.

"And yes, it may be possible running your hardware to avoid any hardware hacking if you are OK with just feeding native smartport telemetry end to end. I really can't be 100% though on this." => I will likely give that a shot when I can, just out of curiosity :) (even if it's not the best setup as you explained)
But thinking about it right now, I don't think it will work as is because the TX module creates S-Port data from analog values passed inside the RC protocol by the RX module. I think it does not take data from the RX's serial RX pin and gateway it to the S-Port output (I'll have to check)

When I have a bit more time in a few weeks, I'll need to do some testing, and try the gitsly branch on openlrsng to see if your telemetry work can be used on existing openlrsng module, since there are so many more of those than the folks using uLRS with hacked up orange modules, and self designed PCBs.
I know people have gotten data to their taranis in the past with the gitsly branch, so I'm hopeful it will work with your teensy converter too.

Best, Marc

athertop, 2017-05-24:
> if you look at that request more closely you'll see it's specifically about the lipo cell tester, which AP does not have, and what I care even more about: RPM sensor and temperature sensors, which AP does not have either (it has some kind of RPM feature, but it does not work for most setups).

Good point Marc - On a purely FrSky setup I use the FLVSS LiPo cell sensors, but of course on my LRS setup (I have only one plane/Wing with this) there is no FrSky telemetry within the model at all, so in this case I do without that sensor, but for my telemetry solution (the one documented in this GitHub repo), I make use of the USE_FLVSS_FAKE_SENSOR_DATA feature (see info here: https://github.com/athertop/MavLink_FrSkySPort/wiki/2.-Options-and-supported-AddOns) which basically takes the Ardupilot battery 1 monitor voltage, and divides it up by the cell count to give me a simulated cell voltage - this is what is displayed then against C-Min on the LUA page. Its a real fudge, but I prefer to check the state of charge by cell voltage rather than pack voltage - saves me having to think, "how many cells is this model?" Your solution of integrating the FrSky sensors into Mavlink directly would be awesome, but that is some challenge, from both a hardware and software perspective. Good luck to you on that one!

> ok, but the mavlink data would still only flow at 19200 anyway, even if you have 57.6k going to the BT module. I suppose you do this because the BT module wants to run at 57.6k, and if so, that's fine.

That's correct, the mavlink data goes across the UHF pipe at 19200 and 57600 at the Rx end is purely to match the connected hardware items as you suggest also.

> I will likely give that a shot when I can, just out of curiosity :) (even if it's not the best setup as you explained)

Cool. Let me know how you get on.

> But thinking about it right now, I don't think it will work as is because the TX module creates S-Port data from analog values passed inside the RC protocol by the RX module. I think it does not take data from the RX's serial RX pin and gateway it to the S-Port output (I'll have to check)

To be honest, I have no idea how your Tx/Rx setup works for passing data through end to end. In the ULRS setup, we of course don't have the built in inverter (in the Tx module), we just have a regular UART port at either end and anything fed in at one end comes out the other in the same way. So its up to the user whether he feeds in Mavlink or Frsky data on that UART. As long as we can invert the output from the Tx end, and implement that diode solution to make a single wire (bi-directional) serial connection, then we can feed that into that SPORT pin in the module bay. This is how I create an inverter/diode setup:
[image]
The RS232<->TTL board is this: http://www.ebay.com/itm/5Pcs-Mini-RS232-To-TTL-MAX3232-Converter-Adaptor-Module-Serial-Port-Board-CL/272491775474
Any old diode will do.

> When I have a bit more time in a few weeks, I'll need to do some testing, and try the gitsly branch on openlrsng to see if your telemetry work can be used on existing openlrsng module, since there are so many more of those than the folks using uLRS with hacked up orange modules, some self designed PCBs, and a (very) few legit looking modules you can buy.
> I know people have gotten data to their taranis in the past with the gitsly branch, so I'm hopeful it will work with your teensy converter too.

You might want to strike up a conversation with geofrancis (on RCG) as he has a ton of experience with OpenLRSNG and also with my teensy solution - he is local to me here in Scotland and is a source of inspiration for me - he is our local drone expert :-)

geofrancis, 2017-05-28:
i have used the teensy solution with gitsly openlrsng successfully, dont get me wrong openlrsng gitsly fork is not the fastest mavlink connection but it has a much longer range than ulrs due to it using a much lower 19200 data rate instead of 115200 and mavlink flow control allowing you to use serial and air rate of 19200 without dropping packets.

openlrsng gitsly gets around 300b/s but its fast enough, the only time you notice is loading parameters or uploading large missions, it might take a minute or 2 but i just usb for that kind of thing. ulrs gets around 1.5kb/s but because of the very air rate it gets poor range on 100mw hardware.

athertop, 2017-05-28:
George, ULRS also has 19200baud data rate. its only the serial connection at the ground side which used 115200 (but now 57600 in recent versions to improve device compatibility). The data is still passing at 19200 though over this. Like your last solution - using teensy with 3dr radios. Never thought of trying this. I never get great range with 3dr radios to be honest. Probably due to badly tuned antennas.

geofrancis, 2017-05-29:
Yes ulrs uses 19200 serial but it's runs the radio at 115200. This let's ulrs get 19200 serial data and RC over the link without dropping packets.

Gitsly uses mavlink flow Control so it can use 57600 serial over a much slower 19200 radio link by dynamically changing the packet rate from the apm so that it can oversubscribe the link without dropping data.

The 19200 radio speed more than doubles the range that's why ulrs has to use 1000mw to get any kind of acceptable range. But people have flow  8k with just 100mw on openlrsng using jus dipole antennas.

geofrancis, 2017-05-29:
With openlrsng you can customise the serial and radio speeds. I could double the range again by dropping the radio to 9600 but with such limited bandwidth I start getting lag in the controls unless I disable telemetry.

athertop, 2017-05-29:
Interesting stuff George. Didn't realise that's how things worked with gitsly. I think flip-flop suggested a future enhancement to allow different data choices, but given the slow schedule for releases we could all be long gone before that gets into the code :-)

geofrancis, 2017-05-29:
Ulrs is a simpler solution and should be compatible with just about any serial telemetry that runs at 19200.

Gitsly has mavlink integration so it communicates with the flight controller to manage data rates and inject rssi. So it only really works with mavlink.

marcmerlin, 2017-05-29:
@geofrancis nice to virtually "run" into you again. Thanks for the details on gitsly, it sounds like it's definitely worth a shot, eventually if the backlink works at reasonable distance on just 100mW.
I'm just a bit worried about the 51 commits that landed in openlrsng after gitsly was forked off, and whether those will affect my somewhat newer brotronics RX/TX modules. I guess I'll have to look at the code and see.
Shame that the project is dead and never got merged :(
I'm not sure when I'll have time to do this, but definitely want to check it out. Will report back here.

geofrancis, 2017-05-29:
I have used gitsly for years and have never had an issue so I wouldn't worry about it, the last few commits were for very minor things like sbus scaling that doesn't get used on gitsly.

geofrancis, 2017-05-29:
The only thing that is really missing is the switch profile selection, it still works for power levels tho

marcmerlin, 2017-05-29:
@geofrancis  thanks, that helps. Profile selection via switch has never really worked well for me anyway, so no big loss :)