athertop/MavLink_FrSkySPort #14: wiki - Setup AP_SYSID

Clooney82, 2017-06-14:
Hi Paul,

while setting up swarming on Missionplanner I changed the parameter **SYSID_THISMAV**, to identify the different copters.
This led to a none working teensy telemetry.
It took me about 2 weeks to figure out why this happend, I almost though my teensy was faulty.

But afterwards the solution was too simple to see it immediatly.
I only needed to change line 121 in MavLink_FrSkySPort.ino
`#define AP_SYSID            1       // autopilot system id
`
to the value of **SYSID_THISMAV**.

Maybe this Information should be added to the wiki on setup page.
If telemetry not working anyone should check if **AP_SYSID** and **SYSID_THISMAV** matches.

BR, Jochen

athertop, 2017-06-14:
Jochen, so how does line 121 now read? And will the change work for all users (even if not involved with swarm setup)?

Clooney82, 2017-06-14:
Hi Paul,

it is not a code change needed. Only as hint for the wiki page.
Default should work out of the box.
It is only in case some users change **SYSID_THISMAV** in ArduCopter/Plane, then they must also change **AP_SYSID** to the same value, otherwise telemetry will not work.
This is because Teensy code will only look for MAVLink message with SYSID 1 (which is default value of SYSID_THISMAV in ArduCopter/Plane)

BR, Jochen

athertop, 2017-06-15:
Jochen, sorry, I read your original message too quickly and didn't take in fully what you meant. I understand now. Yes, I can add this info to the wiki. Cheers for the info. Would be good to hear more about your swarm project.

athertop, 2018-07-09:
Jochen - added this page: https://github.com/athertop/MavLink_FrSkySPort/wiki/2.7---Swarming-and-alternative-Mavlink-System-ID

BR, Paul

closed 2018-07-09 by athertop