athertop/MavLink_FrSkySPort #15: Replace RSSI with RC Channel raw value

Chadwide, 2017-08-13:
Hi Guys, I'm using your project and it's awesome. Thank you. I'm using ULRS and I'd like to replace the Mavlink RSSI value with the RAW value (or a scaled percent like 1000us = 0%) of an RC channel. Is this possible? The reason is that older APM2.6 flight controllers won't read a channels PWM value as RSSI. ULRS is giving me RSSI on CH8 and on mission planner I can just watch this value to see the link's health. I'd love a way to monitor this value from the Taranis. I tried to redefine ap_rssi as mavlink_msg_rc_channels_get_chan8_raw but no luck....

Thanks!

athertop, 2017-10-11:
Chadwide, Sorry, just spotted your issue posted here. Took me a few reads to understand what you mean. To be honest, I'm not really developing the code any further - the code here will work on APM generally but the sounds and AP version on that controller are back at 3.21 so I don't think the effort in adding what you need here would be of significant value to many folks in the community. You can feel free to have a go at modifying the teensy code yourself if you wish. Would suggest looking at Mavlink.ino at line 471 and replace:

 ap_rssi = mavlink_msg_rc_channels_get_rssi(&msg);

with:

ap_rssi = mavlink_msg_rc_channels_get_chan8_raw(&msg)

Make sure line 144 of MavLink_FrSkySPort.ino is un-commented

good luck, Paul

closed 2017-10-11 by athertop

Chadwide, 2017-10-12:
Thanks! I'll give it a shot.