athertop/MavLink_FrSkySPort #16: TxBat showing 0.0v on Telemetry Screen

maroquio, 2017-09-26:
Hi. In the top middle of main telemetry screen, there is a field called TxBat that is supposed to show the Taranis battery level. The problem is that it's showing only 0.0v, all the time. The original initial screen of Taranis shows the Tx voltage, but the athertop telemetry main screen have this problem. Is this a bug or am I missing something? PS: My Taranis is flashed with the latest firmware.

athertop, 2017-10-11:
Hi, I have the latest release of OpenTx (2.2.0) on my Taranis X9D plus, but can not reproduce this issue - the TxBat: value reads correctly on mine. The LUA script simply references the inbuilt telemetry value "tx-voltage". Are you running OpenTx v2.2.0 ?

athertop, 2018-07-09:
No response - closing

closed 2018-07-09 by athertop