athertop/MavLink_FrSkySPort #17: USE_FLVSS_FAKE_SENSOR_DATA doesn't show the single cell value

robustini, 2017-10-14:
The value of the single cell remains at "0.00" on the display, while the total voltage is displayed correctly.
I enable and configure the define:

#define MAXCELLS 3
#define USE_FLVSS_FAKE_SENSOR_DATA

The "Cels" sensor show the right total battery voltage.
Please check, thanks!

athertop, 2017-10-14:
Hey Marco - I replied in the Clooney82 repo also, but better to deal with the issue here, so welcome to my repo!

To address the issue. If you have a FrSky FLVSS sensor in your system monitoring your balance connector, then leave the following lines commented out. If however you do not have one of these sensors, then un-comment the line:

#define USE_FLVSS_FAKE_SENSOR_DATA

as this will cause the code to create an emulated FLVSS sensor, which will appear on the telemetry sensors screen as "Cels".

Either way, following the instruction in the wiki to create a "Cmin" calculated sensor (based on this Cels sensor) is required, as this sensor is what's used in the LUA page. If you have a FLVSS sensor then this will provide the real minimum cell voltage to the LUA telemetry page, if not, and you are using the emulated senor, then it simply divides the Cels value by your cell count. Details in the new wiki here (see section 6 on this link): https://github.com/athertop/MavLink_FrSkySPort/wiki/1.2.-FrSky-Taranis-Telemetry

Cheers, Paul A (athertop)

robustini, 2017-10-14:
Damn, i forgot to create the Cmin sensor, all work fine, thanks!

athertop, 2017-10-14:
Glad you got it sorted Marco!

closed 2017-10-14 by athertop