athertop/MavLink_FrSkySPort #19: rangefinder altitude

geofrancis, 2017-10-14:
i want to program the taranis with terrain warning alarms, so if the rangefinder detects anythign under it closer than say 20m i can make the handset shout "TERRAIN TERRAIN PULL UP"! like a ground proximity alarm on real aircraft. im already using the airspeed for a stick shaker so if the airspeed gets too low it vibrates and shouts "STALL"

i tried using regular altitude but it doesnt change with terrain, it just flashes "ALT VARIANCE" on the screen

athertop, 2017-10-14:
I think it would need two things to make this work. First is that you would need to get the teensy to convert/assign the rangefinder altitude to a s.port sensor - making this available as a sensor in the Taranis. The voice could be setup without LUA, but just with a logical switch and a special function to call the WAV file (and a wav file of course!)

geofrancis, 2017-10-14:
i was thinking replace the Galt sensor with the rangefinder altitude since few people have a need for gps altitude?

athertop, 2017-10-14:
That could definitely be done. Is there a telemetry value which starts as rangefinder value then drops to baro altitude once max range is exceeded? Could use this, would be best to replace AGL Alt value (just shown as Alt) on display with this if it exists in mavlink?

geofrancis, 2017-10-14:
I don't think so,  as far as I know, all the altitude sensors are treated separately

athertop, 2017-10-14:
Found this in the message spec. Not sure which data stream that's in but when I have some time I can maybe have a look at the code. I could give an option to use it instead of Alt - I.e. send rangefinder on the Alt telemetry sensor instead of Baro Alt:
http://mavlink.org/messages/ardupilotmega#RANGEFINDER

geofrancis, 2017-10-14:
alt is useful and needed for the lua whereas Galt is not used for anything so i would prefer to sacrifice Galt over baro altitude.

athertop, 2017-10-14:
Looks like the rangefinder data is included in the EXTRA3 stream. See here: http://ardupilot.org/copter/docs/parameters.html#sr0-extra3-extra-data-type-3-stream-rate-to-ground-station

You're right George, the gAlt we could afford to lose.

athertop, 2017-10-14:
George - Just done the mods to the code in the dev branch give it a go and let me know if it works. Cheers

athertop, 2017-10-14:
Its just the Teensy code which has changed - you'll need to uncomment the line:

//#define  USE_RF_ALT in MavLink_FrSkySPort.ino

geofrancis, 2017-10-14:
nice il try it as soon as i get a chance.

athertop, 2018-07-09:
George, Did you get a chance to test this?? Not planning to continue maintenance on this project so hoping to close off all the issues. Cheers, Paul

closed 2022-09-25 by geofrancis