athertop/MavLink_FrSkySPort #20: RTK Status

geofrancis, 2017-10-15:
i was looking through the lua code and i see it only looks to see if the gps status is over 3 to signify 3d gps lock, 

https://github.com/athertop/MavLink_FrSkySPort/blob/Mav_Sport2.1-2.2-master/Taranis/LUA_Source_files/SCRIPTS/TELEMETRY/LIBRARY/mainrun.lib:257

could that be modified to show mode 4 (RTK float) and mode 5 (RTK fix) also so i can see the status of the RTK

athertop, 2017-10-15:
I guess this can be done, but will need to wait until I have more time. Cheers, P

geofrancis, 2017-10-15:
cool il try just changing 2d to 3d and 3d to rtk fix.

geofrancis, 2017-10-15:
would changing this:

`	if telem_lock >= 3 then
			lcd.drawText (168, 10, "3D",0)
			lcd.drawNumber (195, 10, telem_sats, 0+LEFT)
			lcd.drawText (lcd.getLastPos(), 10, "S", 0)
		elseif telem_lock>1 then
			lcd.drawText (168, 10, "2D", 0)
			lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
			lcd.drawText (lcd.getLastPos(), 10, "S", 0)
		else
			lcd.drawText (168, 10, "NO", 0+BLINK+INVERS)
			lcd.drawText (195, 10, "--S",0)
end`

to this work?

`	if telem_lock >= 5 then
			lcd.drawText (168, 10, "FX",0)
			lcd.drawNumber (195, 10, telem_sats, 0+LEFT)
			lcd.drawText (lcd.getLastPos(), 10, "S", 0)
		elseif telem_lock=4 then
			lcd.drawText (168, 10, "FL", 0)
			lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
			lcd.drawText (lcd.getLastPos(), 10, "S", 0)

                elseif telem_lock=3 then
			lcd.drawText (168, 10, "3D", 0)
			lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
			lcd.drawText (lcd.getLastPos(), 10, "S", 0)

                elseif telem_lock=2 then
			lcd.drawText (168, 10, "2D", 0)
			lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
			lcd.drawText (lcd.getLastPos(), 10, "S", 0)
		else
			lcd.drawText (168, 10, "NO", 0+BLINK+INVERS)
			lcd.drawText (195, 10, "--S",0)
end`

athertop, 2017-10-16:
George, Your suggested code looks spot on! Give it a go and let me know if it works - Actually, would be better if I could compile that luac version for tou to test it. I'll see if I can do that tonight and update the dev repo. Cheers

geofrancis, 2017-10-16:
i think value 4 depends on if your using RTK or not, if using RTK i think 4 = float but with no RTK its DGPS, let me double check.

geofrancis, 2017-10-16:
yes i made a mistake 
https://github.com/ArduPilot/ardupilot/issues/5437

  <enum name="GPS_FIX_TYPE">
              <description>Type of GPS fix</description>
              <entry value="0" name="GPS_FIX_TYPE_NO_GPS">
                  <description>No GPS connected</description>
              </entry>
              <entry value="1" name="GPS_FIX_TYPE_NO_FIX">
                  <description>No position information, GPS is connected</description>
              </entry>
              <entry value="2" name="GPS_FIX_TYPE_2D_FIX">
                  <description>2D position</description>
              </entry>
              <entry value="3" name="GPS_FIX_TYPE_3D_FIX">
                  <description>3D position</description>
              </entry>
              <entry value="4" name="GPS_FIX_TYPE_DGPS">
                  <description>DGPS/SBAS aided 3D position</description>
              </entry>
              <entry value="5" name="GPS_FIX_TYPE_RTK_FLOAT">
                  <description>RTK float, 3D position</description>
              </entry>
              <entry value="6" name="GPS_FIX_TYPE_RTK_FIXED">
                  <description>RTK Fixed, 3D position</description>
              </entry>
              <entry value="7" name="GPS_FIX_TYPE_STATIC">
                  <description>Static fixed, typically used for base stations</description>
              </entry>
        </enum>

geofrancis, 2017-10-16:
let me do a pull request for my edits,

geofrancis, 2017-10-16:
what do you use to edit the lua script?

geofrancis, 2017-10-16:
nevermind im being an idiot trying to open compilled files

athertop, 2017-10-16:
I'll sort it bud, don't need to edit/submit pull etc. You certain on the 2 letter labels you suggested?

geofrancis, 2017-10-16:
```
if telem_lock >=7 then
      lcd.drawText (168, 10, "ST",0)
      lcd.drawNumber (195, 10, telem_sats, 0+LEFT)
      lcd.drawText (lcd.getLastPos(), 10, "S", 0)
  elseif telem_lock=6 then
      lcd.drawText (168, 10, "FX", 0)
      lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
      lcd.drawText (lcd.getLastPos(), 10, "S", 0)
  elseif telem_lock=5 then
          lcd.drawText (168, 10, "FL", 0)
          lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
          lcd.drawText (lcd.getLastPos(), 10, "S", 0)
  elseif telem_lock=4 then
      lcd.drawText (168, 10, "DG", 0)
      lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
      lcd.drawText (lcd.getLastPos(), 10, "S", 0)
  elseif telem_lock=3 then
      lcd.drawText (168, 10, "3D", 0)
      lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
      lcd.drawText (lcd.getLastPos(), 10, "S", 0)
  elseif telem_lock=2 then
      lcd.drawText (168, 10, "2D", 0)
      lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
      lcd.drawText (lcd.getLastPos(), 10, "S", 0)
  else
      lcd.drawText (168, 10, "NO", 0+BLINK+INVERS)
      lcd.drawText (195, 10, "--S",0)

    end
```

geofrancis, 2017-10-16:
I think that should do it

athertop, 2017-10-16:
How does this look instead - less code, and hopefully more efficient:
```
if telem_lock <= 1 then
  lcd.drawText (168, 10, "NO", 0+BLINK+INVERS)
  lcd.drawText (195, 10, "--S",0)
else
  if telem_lock == 2 then lcd.drawText (168, 10, "2D", 0)
  elseif telem_lock == 3 then lcd.drawText (168, 10, "3D", 0)
  elseif telem_lock == 4 then lcd.drawText (168, 10, "DG", 0)
  elseif telem_lock == 5 then lcd.drawText (168, 10, "FL", 0)
  elseif telem_lock == 6 then lcd.drawText (168, 10, "FX", 0)
  elseif telem_lock >= 7 then lcd.drawText (168, 10, "ST",0)
  end
  lcd.drawNumber (195, 10, telem_sats, 0+LEFT )
  lcd.drawText (lcd.getLastPos(), 10, "S", 0)
end
```

athertop, 2017-10-16:
I just did a commit to dev branch, but forgot to do the luac compile of this. Will have to wait until tomorrow as getting late and needs me to boot to Windows to run old version of Companion to do the compile.

geofrancis, 2017-10-17:
i have been playing with custom sensors on the taranis just to see if i could do anything useful and i remembered about feature in minimosd that gave you glide distance if your throttle was at zero 

so it got me thinking about what other virtual sensors that could be made 

glide ratio, Vspeed/airspeed

athertop, 2017-10-17:
George, Just added the compiled version here: https://github.com/athertop/MavLink_FrSkySPort/blob/Mav_Sport2.1-2.2-dev/Taranis/LUA_SD_Card/SCRIPTS/TELEMETRY/LIBRARY/mainrun.lib

so replacing the version on your SD card with this one will allow you to test it - I haven't so please let me know how you get on.

In answer to an earlier question, I edit the files on my Mac using a combo of Github desktop (which gives me a locally synced copy of the files), and an editor called Atom - which is really good as an editor for loads of different languages. I have also used a LUA editor in the past called ZeroBraneStudio (its free and a general LUA editor, available for Windows too). You have to bear in mind though that LUA in OpenTx is a cut-down version, so not all of the LUA libraries are available in OpenTx version. OpenTx LUA manual is here: https://www.gitbook.com/book/opentx/opentx-2-2-lua-reference-guide/details

Cheers, Paul

geofrancis, 2018-03-01:
this works fine shows gps status great.

closed 2018-03-01 by geofrancis