athertop/MavLink_FrSkySPort #22: Status text nonsensensical

fnoop, 2018-01-17:
Hey @athertop 

Just started a plane from scratch and followed your fantastic wiki instructions to get everything working.  I'm running arduplane 3.8.4, opentx 2.2.1 (using your dev branch).

Everything is working perfectly, except that I'm getting mumbo jumbo on the status text screen.  I get this on my older copter setup as well.  The status text used to work really well, is there something I'm doing wrong, or is this a known issue?  I've setup the Fuel/MSG sensor as per the wiki.

I get something like this:
KFIM ITI Y AGNNTOMET T6E2 U1NIALAWLIME CPLNIHO
KFIM TT IGENT LE
KFIMU1ILALNMT MPTE

athertop, 2018-01-17:
Fnoop, it's never worked well for me sadly. It's code written by Clooney82. It appears to be a buffering issue. Feel free to dig in to see if you can sort it. My attention have moved on to real first person flight these days so RC taking a bit if a back seat at moment.

fnoop, 2018-01-17:
Thanks @athertop it used to work reliably so I'll see if I can go back and find where it changed.

athertop, 2018-01-17:
Sometimes it works flawlessly for me then other times it corrupts. Think it needs a better buffering solution.

athertop, 2018-01-17:
Frsky libraries did get updated a while back maybe look towards that. Not aware of changes in the lua.

KhristopherK, 2018-01-31:
Hi @fnoop  where you able to get it working? in my case it's not at all. running apm copter 3.5

fnoop, 2018-02-02:
Nope, doesn't work for me at all :(  It used to work perfectly certainly with my fork and wolkstein, not sure what happened after that.

athertop, 2018-07-09:
Wondering if it got broken after the SmartPort libraries got updated. You guys still using this telemetry project? I have switched to using the yaapu (Alex') passthrough scripts instead (https://github.com/yaapu/FrskyTelemetryScript) and for LRS models I am using this in combo with Eric's teensy project here: https://github.com/zs6buj/MavlinkToPassthru . The messages work great in the yaapu script. Don't think I will be maintaining this code moving forward given these new scripts do the job much better. Your thoughts guys?

fnoop, 2018-07-10:
@athertop Yes I've switched to yaapu as well, it is better on several fronts.  This project still has merit however as it's able to do a few things that passthrough doesn't (such as use the divider sensors, led control etc), but passthrough is pretty compelling.
Thanks for your efforts on maintaining this project over the years, it's been much appreciated.

athertop, 2018-07-10:
Cheers Fnoop! Closing

closed 2018-07-10 by athertop