athertop/MavLink_FrSkySPort #23: teensy setup is not working on Pixracer

ThomasH3, 2018-03-10:
Hi
I'd like to use the same setup on my pixracer but I cannot receive any data. 
teensy wiring and fireware is ok, and telemetry on Pixracer FRS port is working.
Do you have an idea what could be the issues.

athertop, 2018-03-10:
I actually have one quad with a pixracer but never kitted this one out with a teensy (only using FrSky telemetry outputting from UART through an inverter with this one). If I can dig out a spare teensy board then I'll look at giving it a try - think I have an LC board somewhere. It might be a while before I manage to do this though, so if anyone else has this working on Pixracer please feel free to chime in. Cheers

lucamax, 2018-03-11:
I did and I far that I can remember I did not had any problem.

Check twice Pixracer pinout if I remember well there was some error in the documentation.

ThomasH3, 2018-03-11:
I'm using an SD Card.
I can not get Teensy to work, however I can now receive telemetry on FRSky Port.
I was looking to use the "main" Lua script, I really like it on my other APM FC's ....... is it possible to amend it to read Pixracer relemetry ? for example flight modes (Pixracer - tmp2)

athertop, 2018-06-12:
I think your problem may be due to the Pixracer FrSky serial port - it provides data already inverted (TTL), and the teensy code by default expects it as standard RS232 (non-inverted). There may be a way to get around this by amending the code slightly (although I haven't tested this, so try at your own risk):

In Mavlink.ino amend line 62 to read: _MavLinkSerial.begin(_MavLinkSerialBaud, SERIAL_8N1_RXINV_TXINV);

This will change both Tx and Rx pins of the mavlink serial port to expect/provide inverted signals.

If this doesn't work, then sorry I think a hardware inverter (search "RS232<->TTL" for a tiny board from AliExpress) would be required.

closed 2018-06-12 by athertop