athertop/MavLink_FrSkySPort #25: Can't get sensors on taranis.

clemtrinc, 2018-06-11:
I've been trying to make this work on my taranis with an UHF system working as transparent serial.
I'm not familiar with teensy, I flashed it and the behaviour right now is one orange flash when powering the teensy 3.2, then solid orange when powering the aircraft, does it mean that I flashed the teensy correctly and the program is running well ?

Now when I discover sensors, nothing shows up (opentx 2.2.1)

Any help would be hugely appreciated.

athertop, 2018-06-11:
Sounds like you didn't enable polling mode in the teensy code. You need to uncomment (remove //) the line //#define POLLING_ENABLED in MavLink_FrSkySPort.ino

Give that a go please and try discovering sensors again. Cheers

athertop, 2018-07-09:
No response - closing

closed 2018-07-09 by athertop

clemtrinc, 2018-07-09:
Sorry I forgot to update the issue, indeed polling was disabled, works great now thanks !