athertop/MavLink_FrSkySPort #26: How to deselect sensors?

master-of-desaster-88, 2018-07-08:
Hello togeter,
I like to use only a part of the MavLink_FrSkySPort Teensy values.
For example RPM or Airspeed.
The other values I get from the FrSky Passtrough protokoll.
So I like to select and deselect the sensors given from the Teensy.
I got a little bit lost in the code and it would be nice to point me at the right direction.

Thanks and best regards,
Markus

athertop, 2018-07-08:
Markus, I'm trying to figure how you are using the passthrough protocol alongside the teensy running the code in this repo? I take it you have 2 serial ports of the pixhawk configured - one for Mavlink feeding the teensy and another for passthrough, and then feed both these into the s.port connection of the X series Rx?

You may instead want to take a look at a new project, which I consider the next generation from this one, which uses the teensy to provide passthrough alongside a bunch of regular FrSky sensors. You can find it here:  https://github.com/zs6buj/MavlinkToPassthru
Most users of this MavlinkToPassthru project use a LUA telemetry script to display the passthrough data, which you can find here: https://github.com/yaapu/FrskyTelemetryScript. Like I said, you also get a bunch of regular sensors provided as well as the hidden passthrough sensors.

athertop, 2018-07-08:
I should add - if you want to use this teensy code (rather than the new project I suggested above), then to disable sensors you would need to edit FrSkySPortTelemetry.ino, commenting out the sensors you don't want included. Bear in mind that some of the sensors carry multiple pieces of data (only interpreted by the LUA code provided by this project. See the wiki for info.

master-of-desaster-88, 2018-07-09:
Hi Keen, I use the outstanding telemetrie script from Alex (Frsky Telemetry Script). Alex was so kind to create a version for me which I am be able to display four external sensors inside the Horus X12S script. I could toggle the values at the left side of the script.
As a AduroPilot Plane neerd I like to get more informations during the flight without looking to the Mission Planer on the Laptop. For example airspeed versus groudspeed or value of trottle or rc output from the elevator during a FBWA flight.
The most informations I get form the FrSky passthrough protokoll and the yapuu sensors. This is a seperat serial port connected to the S.Bus on the R9 receifer. The missing values I try to get over the MavLink - FrSky passthrough also connected to the S.Bus on the R9 receifer. The Teensy is connected to a own serial port. You were correct, at the moment I use two serial ports at the Pixhawk.
Problem is/was that many ID are double so this will not work.
I will try the new project from Eric as it could solve me problem.
Many thanks for your suggestions an help so far.
Ceers, Markus

athertop, 2018-07-09:
Markus, I think that the zs6buj (Eric's) teensy project should offer you a far better solution - it will also free up a serial port on your flight controller as it will send a mixture of passthrough and standard serial ports through. If you engage Eric he may be willing to add sensors to meet your requirements - you have to be aware though that there are limits as to how much data can be passed through the smartport. I'll close this one just now. Cheers, Paul

closed 2018-07-09 by athertop